Dvogonen/cyclop_plus_plus #3: Picture Bounce

mattyra, 2016-10-24:
When flying on a locked in channel and range becomes extended and rssi becomes low (note 344 in top right) the picture and the OSB begin to jump around and the signal drops and everting goes for a moment before returning....

I am unsure as to the Voltage setting on the RX end as i have no Multi meter at the moment could this effect the matter...

mattyra, 2016-10-25:
Just to update this it seems as tho the channel drops then try to re-tune when RSSI is low

Dvogonen, 2016-10-31:
The RSSI value is the video signal strength. When the signal strength goes lower than the critical level for reception, the video receiver will fail to produce an image. The "jumping around" is the video receiver struggling to hold on to the sync signal of the video it is displaying.
The RSSI value is not an absolute video quality indication. If there is a lot of interference on the channel, the video reception will fail at relatively high RSSI. This will be the case when two transmitters are transmitting on the same channel at the same time.

mattyra, 2016-10-31:
Hello,

no all signals are stand alone during testing, its seems to lose the lock and then fail and black out....

Dvogonen, 2017-01-29:
I am closing this one, since I do not think it is an issue CYCLOP++ has any influence over. The software only tells the receiver which channel to use. From that point on, the receiver works completely autonomously.

closed 2017-01-29 by Dvogonen