Dvogonen/cyclop_plus_plus #15: Grey screen/OSD blinking on medium/low RSSI

superagentorange, 2017-05-10:
Hey, thank you so much for your work on this Project! I did install the the mod without problems and it is working. One problem still remains:
Stability of the video feed has decreased significantly. I get flickering and grey screen/blank screen as soon as RSSI falls a little (i.e. below 400 in the OSD). Before the mod picture would just get a little grainy, now it shuts off completely.

Have you made similar observations? Is it because the minimOSD looses video signal quicker that the installed display driver board?

Hope you have any suggestions on how to improve this, if it can be improved.

Thanks again for your great work!

superagentorange, 2017-05-22:
A small update from me: I couldn't live with the flickering / jumping image with lower reception and reverted the mod for now. Everything went back to normal. I then installed Cyclop+ with a cheap 1306 OLED display and everything is fine now. Thanks again!

nofixedip, 2017-06-09:
I can confirm the symptoms described, however I feel the only serious issue is the grey screen. It's a deal breaker for me. No matter how weak the signal is I still need to see it, in fact the more snow there is the more I need to concentrate on what little remains of the image.

I see the need for the grey screen when using menus in the snow.
Maybe grey screen options, 1 Disabled, 2 Menu screens only, 3 No signal

Aaronsss, 2017-06-10:
Thanks for this software, its a really nice idea and would make the cyclops really good.

I also confirm this issue. In addition to whats been said above I also notice the OSD jumps about half way down the screen sometimes when it flickers. Why would the OSD be going away? Is something resetting the MAX7456 IC? That could explain why it flickers up the screen as if it were turning back on.

However if the signal gets really really bad it does go to the cyclops deault fuzzy no signal screen (no OSD). Could the black screen be occuring only at certain RSSI levels?

I took some DVR footage of it: https://www.youtube.com/watch?v=W9lOaSHYEac

centuryrehab, 2017-06-26:
Just did this mod this weekend. 100% same issue. If i fly with my hyperfloss 5" ultra lightweight that has a whip antenna the screen flickers. :( Wish I didn't switch from Cyclops + to cyclops ++.

Aaronsss, 2017-07-17:
As a work around for this I have put a switch in. One way puts the OSD on the screen the other way runs the video straight from the receiver for flying. It seems to work well until I forget to flick the switch back :)

skaman82, 2018-07-19:
Any progress on this issue? @Dvogonen

skaman82, 2018-07-21:
@Scavanger to bad, I am working on my own GS wit OSD and hit a roadblock recently with the same problem. Unable to solve this yet